newaetech/sonata-pcb #5: 0.2 pcb - User JTAG TCK not routed to clock capable pin

GregAC, 2024-01-03:
The USR_JTAG_TCK net is routed to pin H17 which is not a clock capable pin. Our JTAG TAP implementation uses TCK as a clock (i.e. in a `alway_ff @(posedge tck)` block). As TCK doesn't need to be fast and has little logic connected to it this may not be a problem (we could always refactor the TAP so it doesn't use TCK directly as a clock as well) however it would be preferable for it to be routed to a clock capable pin if possible.

GregAC, 2024-01-03:
@colinoflynn please note

colinoflynn, 2024-01-03:
Oops! Might be another pin-swap problem, will re-route it to a clock pin if reasonable. For now I'd just delegate it to a warning, with the small(ish) logic I probably wouldn't worry about that. I'll get @jpcrypt's input as well once he's available.

GregAC, 2024-01-03:
I've got JTAG over FTDI working so I can load and debug programs on the Ibex demo system with a 10 MHz JTAG clock (or at least that's what OpenOCD claims...)

So a fix for this is in the nice to have bucket but certainly not essential. Could be it's not always reliable at 10 MHz, could try to do some stress testing to determine this. We can always lower the speed if needed.

colinoflynn, 2024-01-03:
Pinswap:
[image]

colinoflynn, 2024-01-03:
Now routed to a SRCC pin (assume SRCC is fine & not MRCC)

colinoflynn, 2024-01-12:
Issue is on 0.3 PCB (now on order) & will be tested once board is built. Issue will be closed once fix is confirmed.

commit 8b9ccafb (2024-01-03): Pinswap to fix #5, TCK routing
closed 2024-02-25 by colinoflynn