newaetech/sonata-pcb #9: 0.2 pcb - LCD_BACKLIGHT pin doesn't go anywhere

colinoflynn, 2024-01-09:
As reported by @GregAC - LCD_BACKLIGHT isn't routed to the FPGA (but does have a single resistor on it so didn't get an ERC when checking).

Easiest is just for the LCD on, by shorting between these pins:
[image]

The relevant schematic section:
[image]

You can also run a bodge wire from the side of R133 that connects to the Q2 gate to an FPGA pin. But in practice the LCD shouldn't need backlight dimming, so just turning it ON is fine.

GregAC, 2024-01-09:
[image]

I just used a big blog of solder to bridge the connections, looks to be working well

[image]

colinoflynn, 2024-01-12:
Issue is fixed on 0.3 PCB (now on order) & will be tested once board is built. Issue will be closed once fix is confirmed.

closed 2024-02-25 by colinoflynn