newaetech/sonata-pcb #10: 0.2 pcb - Level shifter transistor for neopixel is too slow (Q3)

GregAC, 2024-01-12:
I was having issues getting the RGB LEDs to work so took a look with a logic analyser and found a huge delay (~400ns) between the falling edge of the `rgbled0` net and the corresponding rising edge (inverted logic due to shifter) at the DIN pin of the first RGB LED. There was also a smaller delay between the rising edge of  `rgbled0` and the falling edge of the DIN pin.

The timing here is down to the chosen transistor, described in the schematic as a 'DMN3200U' datasheet is here: https://www.diodes.com/assets/Datasheets/ds31188.pdf

The sum of turn on delay and rise time is ~80ns and the sum of turn off delay and rise time is ~570ns. The ws2813b protocol needs timing accurate to ~100ns so this transistor is too slow.

As it happens you can get them to work by taking the delays into account in the hardware generating the signal but the timing of the resulting signal at the DIN is out of spec even if it does manage to still work. Plus given there will be variation across transistors and variation across how out of spec you could go for each ws2813b you may need to 'calibrate' each board separately to get things to work.

A faster transistor would be far better. I happened to have some BSS138 (https://www.diodes.com/assets/Datasheets/BSS138.pdf) on hand so I switched Q3 for one of those and that seems to work fine.

Here's some logic analyzer traces demonstrating the problem

The first shows behaviour with the original transistor, Channel 7 is the FPGA output mirrored to a header pin (and inverted so it matches what should appear on DIN on the RGB LED), Channel 6 is DIN on the first RGB LED. As you can see there is a long time between Channel 7 going high (meaning the `rgbled0` net falls and Channel 6 rising). Channel 6 barely hits the top of its voltage range before dipping again for the next edge.

[image]

After the transistor swap it looks like this:

[image]

Far better rise and fall times.

I did also measure at the actual gate of Q3 (fiddlier as I have to hold the probe in place can't clip it) and that was showing the gate was rapidly charging/discharging so it was definitely the transistor switch time itself at issue, not the time taken to drive the gate high/low.

@alees24 @hcallahan-lowrisc please note

marnovandermaas, 2024-01-12:
Great find Greg!

colinoflynn, 2024-01-12:
Issue is on 0.3 PCB (now on order) & will be tested once board is built. Issue will be closed once fix is confirmed.

closed 2024-02-25 by colinoflynn