newaetech/sonata-pcb #16: Permit store local capability exception

marnovandermaas, 2024-03-22:
In the latest CHERIoT ISA, the store local capability exception is removed. In future revisions of our board, it would be good to remove this LED:
https://github.com/microsoft/cheriot-sail/commit/3d13d24ff974c48897eaf79ac115ac8bbefbc536

colinoflynn, 2024-03-22:
Talk about convenient... that pin can be used to turn off the RGBLED power supply in the next revision then!

marnovandermaas, 2024-05-21:
Actually, this is still a debug feature in CHERIoT Ibex, which can be turned on using a CSR write: https://github.com/lowRISC/sonata-system/blob/6b4fafa9deb16518d98d2627d5e6a3b39cf70103/sw/cheri/error_leds/error.S#L22
Let's leave this in until that is removed.

closed 2024-05-21 by marnovandermaas