newaetech/sonata-pcb #17: KSZ8851SNL missing pull-up resistor for INTRN pin

nbdd0121, 2024-05-08:
The INTRN pin of KSZ8851SNL requires an external 4.7kΩ pull-up resistor which is currently missing.

colinoflynn, 2024-05-08:
Is turning on the FPGA pull-up sufficient for now? We may be able to fit one as a mod on the PCBs that haven't been sent, but it would better if there was a soft solution for now.

nbdd0121, 2024-05-08:
Yes FPGA pull-up seems to be working:

https://github.com/lowRISC/sonata-system/blob/990e491575faec4d297f848c5895e2d54fc07695/data/pins_sonata.xdc#L179

Although I am not sure if this matches the 4.7kΩ requirement.

colinoflynn, 2024-05-08:
Ok phew! The 4.7kOhm should just affect signal speed I'm guessing (e.g., interrupt may be slower on rising edge). We can add the pads at least I think for next PCB spin, but won't do it as a production mod.

colinoflynn, 2024-05-23:
This is now fixed on latest rev (0.9).